rxtx/rxtx #13: rxtx dead?

kkprince1998, 2016-05-17:
Is RxTx Dead?

stroucki, 2017-05-18:
Do you know of an alternative?

hurynovich, 2017-05-18:
Try https://github.com/scream3r/java-simple-serial-connector

stroucki, 2017-05-18:
Interesting, thanks. What is the origin of your library, and are you still maintaining it? How does one migrate from rxtx?

hurynovich, 2017-05-18:
I am not founder of this lib but just user.
This project was not updated since 2014, but seems like it is still best lib to communicate via COM for Java.
I am not really familiar with RxTx just tried it and rejected it in favor JSSC so cannot help with migration.

GallusMax, 2018-10-21:
After rxtx.org seems gone, some history and archives can still be found
here: http://rxtx.qbang.org/
and here: http://fizzed.com/oss/rxtx-for-java

bruce-stewart, 2018-11-16:
I've been very happy with http://fazecast.github.io/jSerialComm recently. One of the very useful features is that all of the different native libraries are bundled in the jar artifact and are loaded automatically. Basically, you only need to add the jar file as a dependency in your project.

nicola-lunghi, 2019-03-04:
@hurynovich 
Hi thanks
the problem with this repo then is that you "stole" the name of the official library
so now everyone is coming here to submit patches/bugfixes

Also the code into the repo is not from the original package.

Can you please create a master branch with the original code so people can start contributing on it to fix bugs /issues?

the rxtx is still a widely used library everywhere.

Also I think you should contact the original author Trent Jarvi tjarvi@qbang.org

Cheers
Nick

alibabashack, 2019-03-04:
Hi @nicola-lunghi,
he did not steal anything. Also, I did not steal anything, not even in a figurative sense. The only thing this repository changed in regard to the original provided by Trent Jarvi is support for the maven build system. The original lib lacked a reproducible cross platform build and a loader for the native libs, which I added here. You're welcome. The rxtx github organization was created with the best intentions, to revive the lib.

> everyone is coming here to submit patches/bugfixes

I had to laugh about this one. Please take a look how many patches and bugfixes were provided. Then take a look when this repository was created. It would be more accurate to say, that nobody cared about anything here.

Sadyl, I did not have time in the past to maintain the requested changes. If somebody wants to take over maintenance, please perform a review of the pull requests received so far, add tests, etc. Your work is welcome.

Regards Alex

mizapf, 2020-09-08:
Hi Alex (we had some E-mail exchange on January 1, 2018),

I cannot build RXTX with the current JDKs anymore because the maven-native plugin relies on the javah command, which has been removed from the JDK (>8). Is there anything that can be done about that in the Maven files of RXTX, or do we depend on the authors of the maven-native plugin?

Greetings,
Michael

stroucki, 2020-09-08:
Is "javac -h" equivalent to "javah"?

alibabashack, 2020-09-08:
@mizapf 
Hi Michael,
to be honest I forgot the details of this project over the years and have not looked at the java world for a long time. Please take a look at the github pages documentation at http://rxtx.github.io/rxtx/developers/build/intro/overview.html for the general build flow. It seems to me the process of header file generation by javah is controlled from within the maven-native plugin. If this plugin is still alive there should be a workaround provided by it's authors, I guess.

Forgive me for not being able to help in more detail.

Greetings,
Alex

mizapf, 2020-09-08:
@stroucki 
As far as I heard, javac -h took the role of javah. However, javah is invoked by the maven-native plugin, and it defines specific javah tags; I don't know how to fix that. Eventually, this may be an issue for mojohaus who wrote the maven plugin, but before I'm involving another party, maybe someone here has an idea how to substitute javah. 

@alibabashack 
Hi Alexander, no need to apologize; two years ago I started looking at the current RXTX, but lost it out of sight as well. I was triggered again some days ago when someone tried to get my program running that uses RXTX, and we discovered that I already checked in my changes that relied on the new RXTX implementation (with DriverManager). I'd be happy to help making it build and run again.

stroucki, 2020-09-08:
I just spent a couple minutes hacking on it and I couldn't get the correct maven incantation; I'd assume this is an issue that should be handled by the native-maven-plugin.

Perhaps you can create a wrapper shell script to javac -h, call it javah, and make it available in the PATH.

mizapf, 2020-09-09:
If javah is not needed but javac -h, could we use another maven plugin (the one for javac) to do the job of maven-native? 

I already thought about writing a wrapper script, but some seconds later that idea did not look that good any more. I mean, I can do it, but who else? We surely don't want to write such a recipe into the INSTALL file. Right now, people will only be able to build RXTX if they find an old JDK 7 in some archive.

So if there is no way around maven-native, I'll file a report at mojohaus and hope they fix their plugin.

alibabashack, 2021-04-05:
Hi everyone, hi @mizapf ,
I just migrated the build process from deprecated/removed javah to javac -h and merged some of the long standing pull requests. Could you please give it a try?

I am also motivated to finally get a binary delivery process running. Help is much appreciated. Anyone with knowledge of Github Actions CI?

The confusion of people coming to this repository is unfortunate but understandable. I did not do a good job in communicating the state of the project and the motivation to create this repository.

tl;dr; A few years back the original rxtx.qbang.org was slowly dying. The things that annoyed me the most, were the old CVS and the complicated and mostly manual build process to create a release. So  I migrated the sources of the original rxtx.qbang.org repository to mercurial and later git and with the help of @tuxedo0801 created a one stop maven build. This was a process communicated with the rxtx mailing list. It was not intended as a fork, but as a new edition of rxtx. Sadly the whole mission fell asleep after a while, because nobody seemed to be particularly interested in rxtx as a project. Everybody seemed to be using some ancient binary version downloaded somewhere, without caring to much how these binaries were built.

But, yes, this repository is the successor of rxtx.qbang.org. And if you care about rxtx, please help. It is much appreciated.

There is a clear mission here:
* Finally, get the (already working) build running on some CI and deliver binaries for all major platforms and provide them for download/on maven central. This closes the feedback loop to our users, who are not typically building from source. Especially not Windows users.
* Update the documentation and http://rxtx.github.io/rxtx/
* Discuss required API changes.
* Cleanup API and implementation with respect to clean code

Best regards,
Alex

mizapf, 2021-04-05:
Seems to build now, using openjdk 11.0.10. 
mvn package creates several jars; would it be possible to create a single jar with all required native libs and Java classes? (Or one such JAR per operating system?)

alibabashack, 2021-04-05:
@mizapf The idea is that rxtxSerial contains the service provider implementation and all binaries. rxtx-api is a separate artifact which provides the public API. This forces user to code against the API only, whereas the implementation is a runtime dependency only. All other artifacts are intermediate build products.

mizapf, 2021-04-05:
This means at runtime you have to include both API JAR and the serial JAR in the classpath, right?

alibabashack, 2021-04-05:
As of https://maven.apache.org/guides/introduction/introduction-to-dependency-mechanism.html the API would be in compile scope and the implementation in runtime scope. Both will be at the classpath at runtime, but only the API is in the compile time transitive classpath.

tuxedo0801, 2021-04-06:
Nice to see that this makes some progress.

Just for recerence as @alibabashack mentioned me... there was or is still the rxtx rebundled package around: https://gitlab.com/alexanderchristian/rxtx-rebundled

Did not use it for years, might still work. 

Serial communicarion is heavily used for those wh use the Arduino IDE... I wonder what they use to do the RS232 communication...

alibabashack, 2021-04-17:
Thank you @tuxedo0801 for reminding, but I hope with the upcomming CI/CD workflow we can finally switch to up to date binaries. CI/CD is already working and currently available for review in pull request #32. Feedback welcome!

This means we have new up to date binaries built via github actions which should run on the following platforms:
* linux: x86, x86_64
* windows: x86, x86_64
* osx: x86_64

With very little work also linux arm (armel, armhf) might work.

Since I only have access to linux/x86_64 machines I am looking for volunteers who could run a simple loopback test on other platforms. Also feedback is welcome regarding platform requirements. Where else do you need to run rxtx? ppc mac, 64bit arm, etc.?

EliasIEE, 2021-05-18:
Hi @alibabashack I am very happy to see that rxtx is not dead after all. We were still using a very old library, but since there were no binaries available for Centos 8, we started to switch to another library (which was not very successful so far). Is there already a binary for linux X86_64 that we can test?

EliasIEE, 2021-05-18:
We were now able to create the binary ourselves using the instructions with maven, which worked fine so far. Next step is to test if the communication works...

alibabashack, 2021-05-18:
Hi @EliasIEE, yes, there is a github action CI, now. Its build includes the binaries for linux (x86, x86_64), windows (x86, x86_64), osx:
https://github.com/rxtx/rxtx/actions/runs/844341221 

The artifact _rxtx-distribution_ includes both relevant jars. For details see https://github.com/rxtx/rxtx/issues/24#issuecomment-817287239

EliasIEE, 2021-05-18:
That's great thanks a lot!

telcotronics, 2023-09-19:
No se si estoy loco pero ley toda la historia apasionante de como este tipo cambio la historia.. y el mundo aun no lo conoce..
a mis 40 años vuelvo a JAVA.