rxtx/rxtx #14: error C2011: 'timespec': 'struct' type redefinition, in Visual Studio 2015

ML1, 2016-11-08:
Hello,

I got  error C2011: 'timespec': 'struct' type redefinition, in Visual Studio 2015 error when compiling on Visual Studio 2015.  I already defined HAVE_STRUCT_TIMESPEC preprocessor to prevent error but it doesn't solved it. What will be the problem?

Regards

alibabashack, 2021-04-05:
Hi @ML1 ,
we currently do not support building with Visual Studio. Our Windows build is a cross build on Linux. I know you asked this ages ago, but what is the build setup you are using? We could add support for Visual Studio in the maven build if you think it is useful. To make this happen I need installation instructions and build command lines including compiler names and build flags. Do you think you could provide these?

Best regards,
Alex

alibabashack, 2021-05-15:
I think this was fixed in #32 . Please reopen if the problem persists.

closed 2021-05-15 by alibabashack