rxtx/rxtx #18: HUPCL is unconditionally cleared in configure_port

a3f, 2019-12-09:
The Linux default is that new ports have the HUPCL flag (hang up) set, so RTS is released when the last user of the port closes the device file. rxtx always reconfigures new ports with

```
ttyset.c_cflag = CREAD | CS8 | CLOCAL;
```

which overrides the HUPCL flag. Here is an example how this breaks HW that expects RTS to be inactive when the serial port is unused:
https://forum.arduino.cc/index.php?topic=46144.0

I've a similar setup where a modem expects RTS to be deasserted when the host doesn't communicate with it. This needs to work even if the application terminates
abnormally, so I use the same fix locally, but would like to have the functionality upstream as well.

As I guess there might be *nix systems where HUPCL is not a default, I could draft a patch that doesn't remove HUPCL if it's already set for a port. This would still break setups, where the user runs rxtx, sets RTS to active, closee the application and expects the RTS to stay asserted. I don't know how common such a use case is though.

Thoughts?

christianlupus, 2019-12-10:
> As I guess there might be *nix systems where HUPCL is not a default, I could draft a patch that doesn't remove HUPCL if it's already set for a port. This would still break setups, where the user [...]

I would assume the current setting might be used by a few projects. Changing this behaviour might break these projects.

How about the possibility to enable this HUPCL as you desire depending on an optional parameter in the open function? Then existing code continnues to work but you can use your HUPCL flag as desired.

a3f, 2020-02-21:
> How about the possibility to enable this HUPCL as you desire depending on an optional parameter in the open function? Then existing code continnues to work but you can use your HUPCL flag as desired.

Sounds good. I'll forward the feedback and close the issue for now. If we implement it on top of this fork, I'll open a pull request.

Cheers,

closed 2020-02-21 by a3f