rxtx/rxtx #20: OpenJDK 11 JVM Crash

heywiki, 2020-05-13:
I am getting this JVM crash on OpenJDK 11 when using rxtx on Windows 10

```
COMPortTracer listening on 'COM4' with baudRate '9600', databits: '8', stopbits: '1', parity: '0', sleeptime: '300', flowControlMode: '-1'...
WARNING:  RXTX Version mismatch
        Jar version = RXTX-2.2pre1
        native lib Version = RXTX-2.2pre2
COMPortReceiver COM4 START at 2020-05-13 07:33:49, baud:9600, sleep:300...
Mai 13, 2020 7:33:49 VORM. org.nuveon.device.port.COMPortReceiver run
INFORMATION: COMPortReceiver COM4 START at 2020-05-13 07:33:49, baud:9600, sleep:300...
#
# A fatal error has been detected by the Java Runtime Environment:
#
#  EXCEPTION_ACCESS_VIOLATION (0xc0000005) at pc=0x0000000180003990, pid=12268, tid=5824
#
# JRE version: OpenJDK Runtime Environment (11.0.7+10) (build 11.0.7+10)
# Java VM: OpenJDK 64-Bit Server VM (11.0.7+10, mixed mode, tiered, compressed oops, g1 gc, windows-amd64)
# Problematic frame:
# C  [rxtxSerial.dll+0x3990]
#
# No core dump will be written. Minidumps are not enabled by default on client versions of Windows
#
# An error report file with more information is saved as:
# C:\home\wiki\device\hs_err_pid12268.log
#
# If you would like to submit a bug report, please visit:
#   https://github.com/AdoptOpenJDK/openjdk-support/issues
# The crash happened outside the Java Virtual Machine in native code.
# See problematic frame for where to report the bug.
#
```

[BraysTerminal](https://sites.google.com/site/terminalbpp/) seems to have no problems reading the values 

Here's the code i am using (Snippet)

```
byte[] buffer = new byte[1024];
int len = -1;
InputStream in = m_serialPort.getInputStream();

while (!Thread.currentThread().isInterrupted() && m_run) {
    if ((in.available()) > 0)  {
        len = in.read(buffer);
        if (len > -1) 
        {
            // only notify observers if there is something in the buffer
            if (len > 0) {
                String codeBuffer = new String(buffer, 0, len);
                this.notifyObservers(codeBuffer);
            }
        }
    }
    
    Thread.sleep(m_sleeptime);
}
```

sudhirbitsgoa, 2020-12-29:
Same issue from my end as well.

clackmannan, 2021-03-09:
This is fixed in NrJavaSerial, any idea what has to happen to get the fix implemented in rxtx?
https://github.com/NeuronRobotics/nrjavaserial/issues/131

christianlupus, 2021-03-22:
The most commits did @alibabashack. Maybe he knows if merging is possible somehow.

alibabashack, 2021-03-23:
Hi everyone, I will try to allocate some time next weekend. It will probably take me a while to set up all the cross compiler toolchain infrastructure again. I haven't done any Java projects for several years now (since JVM 1.6 I believe), so my motivation to work on RXTX dropped over time. But now I feel curious how things have evolved and I am happy at least some people still seem to be using RXTX.

alibabashack, 2021-04-05:
I started investigating how the nrjavaserial fork handles this issue.

The first commit of their fix https://github.com/NeuronRobotics/nrjavaserial/pull/133/commits/992153d52d693c459ea8ebf3d7762ddcb59c34ba does not apply to our code base, because our return type of get_java_var_long() is still jlong. This type is defined in the JDKs jni_md.h. It seems they had changed it to long for some reason, which is too small to hold a jlong on some platforms.

alibabashack, 2021-04-05:
Based on the mentioned version string RXTX-2.2pre2, it seems @heywiki, @sudhirbitsgoa, @RaspCla are using an old binary distribution of rxtx. We never released any public binaries since the switch to the maven build system. That's unfortunate, because the main motivation for this switch to a transparent build process was the aim for a continuous delivery of binaries. I think, the best way to serve users is to finally get a continuous integration flow to work. This is the only way to get Windows users in a feedback loop, because they cannot currently build their own binaries. I will try to come up with something. Help much appreciated.

RaspCla, 2021-04-05:
I'm willing to help if I can support. But I don't have any experience with java and maven, yet. Is there anything I can do?
Availalbe systems are:
- Windows10
- Ubuntu 19.10 (virtual machine)
- Linux Mint (most current version) on an own machine

alibabashack, 2021-04-05:
@RaspCla Thanks a lot for volunteering. I have added some task issues with ideas. Considering your post I think you are the perfect candidate for https://github.com/rxtx/rxtx/issues/24. Maybe you can start to get familiar with the build process on Ubuntu. Then maybe make it work on Mint, too and eventually perform the cross build for windows and try it out on your Windows machine. That would be huge! Please feel free to assign this or any other ticket to yourself.

alibabashack, 2021-08-06:
Hi @heywiki ,
could you please try again with the latest binaries from our "rxtx-distribution" github action CI: https://github.com/rxtx/rxtx/actions/runs/844341221

PrasadReddy75, 2021-09-15:
A fatal error has been detected by the Java Runtime Environment: # SIGSEGV (0xb) at pc=0x00000000000039fe

We have a debian package called thirdparty and while de packaging this debian, intermittently, the server is crashing. We are using java 8.

Following is the Problematic frame
C 0x00000000000039fe
And complete error dump is here
[hs_err_pid1553.log](https://github.com/rxtx/rxtx/files/7173393/hs_err_pid1553.log)

It appears the problematic is with RXTXPort calls ? And I'm not sure how to resolve it. Any help would be greatly appreciated.

{code}
0x00000000000039de:
[error occurred during error reporting (printing registers, top of stack, instructions near pc), id 0xb]

Register to memory mapping:

RAX=0x0000000000000000 is an unknown value
RBX=0x00007fc3bc5b0898 is pointing into the stack for thread: 0x00007fc3d0ef5000
RCX=0x00007fc3d7f4030d: nanosleep+0x2d in /lib/x86_64-linux-gnu/libc.so.6 at 0x00007fc3d7e74000
RDX=0x0000000000000000 is an unknown value
RSP=0x00007fc3bc5b07a8 is pointing into the stack for thread: 0x00007fc3d0ef5000
RBP=0x00007fc3bc5b0918 is pointing into the stack for thread: 0x00007fc3d0ef5000
RSI=0x0000000000000000 is an unknown value
RDI=0x00007fc3bc5b07c0 is pointing into the stack for thread: 0x00007fc3d0ef5000
R8 =0x0000000000000005 is an unknown value
R9 =0x0000000000000006 is an unknown value
R10=0x0000000000000006 is an unknown value
R11=0x0000000000000000 is an unknown value
R12=0x00007fc3bc5b07c0 is pointing into the stack for thread: 0x00007fc3d0ef5000
R13={method} {0x00007fc366af3d90} 'eventLoop' '()V' in 'gnu/io/RXTXPort'
R14=0x00007fc3bc5b0a10 is pointing into the stack for thread: 0x00007fc3d0ef5000
R15=0x00007fc3d0ef5000 is a thread

Stack: [0x00007fc3bc4b1000,0x00007fc3bc5b2000], sp=0x00007fc3bc5b07a8, free space=1021k
Native frames: (J=compiled Java code, j=interpreted, Vv=VM code, C=native code)
C 0x00000000000039fe
C 0x00000000000003e3

Java frames: (J=compiled Java code, j=interpreted, Vv=VM code)
j gnu.io.RXTXPort.eventLoop()V+0
j gnu.io.RXTXPort$MonitorThread.run()V+12
v ~StubRoutines::call_stub
{code}

Thanks,
Prasad

alibabashack, 2021-09-16:
Hi Prasad,
could you please try again with our binaries as suggested in my previous post?

Please note that there was a small change in the API which requires one more call during application startup in order to load the native libs. See https://github.com/rxtx/rxtx/issues/24#issuecomment-817287239

PrasadReddy75, 2021-09-16:
Thank you very much for the response. Can you please guide from where I can download the latest libs?
    rxtx-api/target/rxtx-api-2.2-stabilize-SNAPSHOT.jar (contains the java API only)
    rxtxSerial/target/rxtxSerial-2.2-stabilize-SNAPSHOT.jar (contains the Java serial driver implementation and its native libraries)
    to the projects netbeans project.

Thanks,
Prasad

alibabashack, 2021-09-16:
I just realized the github action results expire and theses binaries are no longer available for download here. :( If you can find out a way how to trigger a github action again without new commits I would rerun it. Atm I don't know how to do that and I have no time till Monday for Investition, sorry.

PrasadReddy75, 2021-09-16:
Thanks for the update. Let me check with @RaspCla, in case if he has a copy those libs, if not, please respond to me by Monday with active links to download.

Dear @RaspCla,

Right now it appears the links to download rxtx-api-2.2-stabilize-SNAPSHOT.jar and rxtxSerial-2.2-stabilize-SNAPSHOT.jar have been expired. If you have a copy of them, can you please share here, so that I can download.

Thanks,
Prasad

RaspCla, 2021-09-16:
@PrasadReddy75:
I only own the versions from April 12th, 2021.
Here you can find a copy for a while:
https://cloud.web.de/ngcloud/external?guestToken=KmJhrWSYTOuIp5ZR7I7-ow&loginName=claus.moessner

Claus

PrasadReddy75, 2021-09-16:
Great, and I believe it should work! Thank you, Claus. Much appreciated.
I'll correct my classpath with these updated bins/libs and post my update soon. 
Thanks again.

PrasadReddy75, 2021-09-16:
I deleted the old rxtx binaries and downloaded rxtx-api-2.2-stabilize-SNAPSHOT.jar and rxtxSerial-2.2-stabilize-SNAPSHOT.jar and kept in the class path.
Some of my srs files complaining '**The type RXTXPort is not visible**'
I see this RXTXPort class under rxtxSerial-2.2-stabilize-SNAPSHOT\gnu\io\impl\serial\RXRPort.class, but unable to import in my eclipse.
Can you please advise if I'm missing any thing else?

migry, 2021-12-17:
Apologies, but I am very confused. I cannot find any source code or DLLs for Windows in this repository.

I am trying to compile some old Java code which uses the serial port via a FDTI USB converter. I know very little about Java, so I have spent quite a few hours trying to get the old Java code to compile (there were no instructions). Now I can finally run the GUI as soon as the program uses the USB serial board, I see a TX (or RX) LED flash, and I get a crash. Looks similar to the one reported in this issue.

Note: I am running WIn10 64 bit, and compiled using JDK 1.8.0_311 64 bit (pretty recent I believe). I found the 64 bit RXTX DLL and JAR on the "fizzed" website.

I think I need a 64 bit rxtxSerial.dll which fixes whatever bug is causing the crash in this DLL.

```
/cygdrive/g/MISC/SUMP/LogicAnalyzer/V0.4/client > java -jar analyzer.jar
COM9
Capture...
Rate: 200000000
Size: 2048
Ratio: 1.0
00010001 00000000 00000000 00000000 00000000
11000011 00000000 00000000 00000000 00000000
11000101 00000000 00000000 00000000 00000000
10000011 00000000 00000000 00000000 00000000
10001001 00000001 00000000 00000000 00000000
10100101 11111111 00000000 11111111 00000000
#
# A fatal error has been detected by the Java Runtime Environment:
#
#  EXCEPTION_ACCESS_VIOLATION (0xc0000005) at pc=0x0000000180005b00, pid=1808, tid=0x0000000000005a5c
#
# JRE version: Java(TM) SE Runtime Environment (8.0_311-b11) (build 1.8.0_311-b11)
# Java VM: Java HotSpot(TM) 64-Bit Server VM (25.311-b11 mixed mode windows-amd64 compressed oops)
# Problematic frame:
# C  [rxtxSerial.dll+0x5b00]
#
# Failed to write core dump. Minidumps are not enabled by default on client versions of Windows
#
# An error report file with more information is saved as:
# G:\MISC\SUMP\LogicAnalyzer\V0.4\client\hs_err_pid1808.log
#
```

I found the RXTXcomm.jar and rxtxSerial.dll on the "fizzed" repository which appears quite old and possibly no longer supported, judging by the dates. I am unsure how that website links to this repository.

Some googling suggested a mis-match of parameter types between the DLL and the Oracle JDK

A debug file is created with lots of hex data, but I have no idea how to interpret it!

[hs_err_pid21400.log](https://github.com/rxtx/rxtx/files/7736336/hs_err_pid21400.log)

RaspCla, 2021-12-17:
@migry
the source code you can find here:
[https://github.com/rxtx/rxtx](url)

The build instructions are here:
[http://rxtx.github.io/rxtx/developers/build/intro/overview.html](url)

My communication with alibabashack (while I was performing the build) you will find here:
[https://github.com/rxtx/rxtx/issues/24](url)

And the DLL's you can find within the link to my repository:

RaspCla, 2021-12-17:
@PrasadReddy75
I'm very sorry but I didn't get a message after your reply to my post from september
Nevertheless I'm neither an JAVA nor eclipse expert and my knowledge about the RXTX api is very limited.
Maybe @alibabashack can support

migry, 2021-12-20:
@RaspCla thank you for the information.
WWhat is strange is that the repository appears to be almost empty. There is some java stuff, but many folders just have a "pom.xml" file, whatever that is.
The link (above) to the DLL is not working for me.

For the time being I've had to use my old PC. I will try to install the same versions of software (Java, DLL's, etc) on the new machine.
Unfortunately I do not know how to interpret the core dump in the DLL, let alone know how to fix it!
A group which generated the 64 bit DLL appears to have gone quiet in 2015, with no more posts to their blog/website or Twitter account.

RaspCla, 2021-12-20:
Hello,
sorry, i found out that I provided not working links, I corrected the links above.
Yes, you are correct, there are only some java stuff and only some xml files. But these was the repository I translated as explained by alibabashack. He was the one who fixed the bug with crashing DLLs, 

Within my project I just had to include both Lib files and the crash was gone. I didn't had to copy the dll files directly.

If you like to see how the lib is implemented in within the prm80-project, take a look to:
[prm80x0/Controler](https://github.com/RaspCla/prm80-control-software/tree/main/control_software/trunk/src/f4fez/prm80x0/Controler)
(serialxyz.java)

alibabashack, 2021-12-20:
Hi everyone, @RaspCla @migry 
I am sorry this project is somehow still a mess, but I've been trying to make things better this weekend and worked on the github actions release mechanism. There is still a small problem with versioning, so I have not created a pull request yet. Until there is an official release, you may use the following snapshot release for your experiments:
[experimental snapshot binaries](https://github.com/alibabashack/rxtx/releases/tag/v2.2.1)
You need to add both .jars to your project.

Please take a look at https://github.com/rxtx/rxtx/issues/26 for usage. Note 
`DriverManager.getInstance().loadDrivers();` is crucial. It selects the correct binary libary for your system, extracts and loads it. The binary libs are contained within the rxtxSerial*.jar (you can extract .jars using your .zip tool, they are just .zip files to see what is inside.)

> WWhat is strange is that the repository appears to be almost empty. There is some java stuff, but many folders just have a "pom.xml" file, whatever that is.

Yes, I am afraid this is the structure enforced by the maven build system. pom.xml is to maven what a Makefile is to make. There are so many targets, because every OS requires its own binary lib. They are all build from sources in _rxtxSerial-native/src/main/c_ This is where Java's _compile once run everywhere_ strategy clashes with the C-World. Maven cannot project this setting like e.g. CMake would. But now, I try to finish this approach first, before we may optimize these issues.

It is really nice to see you are helping out eachother and help to build a community around rxtx again. Sorry for the lack of information. It may be frustrating, but I try to help out, step by step.

Best regards, Alex

AloisHofstaetter, 2021-12-22:
Hi everyone!

I **was** having the same problem as described above.
With Linux it worked very well.
In Windows 10 it was not working at all.
Playing around with different JDKs/JREs did not help at all.

This
> [experimental snapshot binaries](https://github.com/alibabashack/rxtx/releases/tag/v2.2.1) 

and

> `DriverManager.getInstance().loadDrivers();`

did the trick.
It works perfect for me.
(Linux 5.10.0-9-amd64 and Windows 10 64 within Virtualbox)

Thank you very much!
Best regards, Alois

davutg, 2022-10-27:
> Hi everyone!
>
> I **was** having the same problem as described above. With Linux it worked very well. In Windows 10 it was not working at all. Playing around with different JDKs/JREs did not help at [...]

The link provided for experimental snapshots is broken. I'm trying to make it work for Microsoft JDK 17. Created headers manually by javac -h  https://github.com/microsoft/openjdk/issues/330#issuecomment-1293412144 but still couldn't manage it work. Don't know exactly what are the requirements to build this project manually apart from mingw64 builder. I went thru bash script it throw below error. It looks lsb (something like linux subsystem base which is also used in docker WSL I believe) is missing.. 

x86_64-w64-mingw32-gcc-12.2.0 command -v lsb_release

C:/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/12.2.0/../../../../x86_64-w64-mingw32/bin/ld.exe: cannot find lsb_release:

davutg, 2022-10-30:
This branch works with MS JDK 17 as well https://github.com/NeuronRobotics/nrjavaserial/

rafa9811, 2022-12-15:
@davutg did you get something? thanks!

davutg, 2022-12-15:
@rafa9811 As mentioned this is working https://github.com/NeuronRobotics/nrjavaserial/ 
as expected, find the release here
https://github.com/NeuronRobotics/nrjavaserial/releases/tag/5.2.1

ThomasW2005, 2024-03-03:
Is there still any work being done here? Because i still get the same error