rxtx/rxtx #24: Validate Build Instructions

alibabashack, 2021-04-05:
Our current build instructions can be found at http://rxtx.github.io/rxtx/ . It is very important these instructions are clear, easy to understand und up to date to everyone interested in working on rxtx. This is where even novices can shine, because they can identify missing parts. So if you want to help please pick a challenge below and report if there is anything you do not understand or that does not work. Don't hesitate to be picky. The problem is not you, it is bad documentation. We are open for your suggestions.

* Read the build process overview. What is the purpose of each of the artifacts? Asks questions here if unsure.
* Perform a development build only including native binaries for your machine on your Linux distribution of choice. Pick the best matching available build platform instructions as a template. Then hack your way through and make notes whenever you have to deviate from the instructions (e.g. different package names for the same thing, etc.).
* Can you find the resulting jar containing the native library? Write a quick example program using it. You did? Great, please donate it as an example for documentation or as a smoke test.
* Now, also add cross builds for other binaries, e.g. for Windows.
* Let your example program run on that other target platform and check if it still works.
* Try to perform a release build as described. You might need other people who can provide the apple binaries.
* Add a new page for your build platform to http://rxtx.github.io/rxtx/ (gh-pages).

RaspCla, 2021-04-06:
How can I assign myself to this task? (I'm just beginner level at github, also)

alibabashack, 2021-04-06:
Sorry, I think this is a permission problem and you actually can't assign yourself at this time.

alibabashack, 2021-04-06:
Nice, thank you for your time. This is good feedback already and tells me the documentation of the build is not too bad, but there is absolutely nothing that covers usage, yet. I think in order to not waste your time I will have to come up with a least some documentation in this direction first. This won't happen any time before Friday I am afraid.

Very motivating to see another project depending on rxtx :) I hope we will make it work again. I'll come back soon with the required documentation.

RaspCla, 2021-04-08:
Yes, the documentation is really good. 
I had to fight with PATH and environment variables, but google helps a lot (I'm more experienced with Windows than Linux).
The onliest thing you could add to the documentation ist, that the "mingw-w64" or similar package is needed for windows.
The comment: "various cross compilers" within the build instruction didn't help to understand what to install (I was guessing which tool is missing after reading the log/error messages :-)

alibabashack, 2021-04-11:
Hey, since it might take some time for me to update the docs, here the short version. It might allow you to get your project running:
After building rxtx for the platforms you target, add
* rxtx-api/target/rxtx-api-2.2-stabilize-SNAPSHOT.jar (contains the java API only)
* rxtxSerial/target/rxtxSerial-2.2-stabilize-SNAPSHOT.jar (contains the Java serial driver implementation and its native libraries)
to the projects netbeans project.

Then add the following line somewhere in your project. It must be executed once before an attempt to open a serial port is made:
`DriverManager.getInstance().loadDrivers();`
This will extract and load the appropriate native library from the jar. There is no need to copy the dll manually.

I wasn't able to make the PRM80X0 project build with netbeans 12.3 and try it out, because some other dependencies are missing. But the rxtx part looked fine.

RaspCla, 2021-04-11:
Hi,
I tried everything you stated above, but without success.
I always see following error message, independendly where I add the "DriverManger.." line
`Exception in thread "AWT-EventQueue-0" java.lang.IllegalArgumentException: setSelectedIndex: 0 out of bounds
	at java.desktop/javax.swing.JComboBox.setSelectedIndex(JComboBox.java:622)
	at f4fez.prm80x0.gui.OptionsDialog.<init>(OptionsDialog.java:39)
	at f4fez.prm80x0.gui.PRM80X0View.optionMenuItemActionPerformed(PRM80X0View.java:774)
	at f4fez.prm80x0.gui.PRM80X0View$26.actionPerformed(PRM80X0View.java:578)
	at java.desktop/javax.swing.AbstractButton.fireActionPerformed(AbstractButton.java:1967)
	at java.desktop/javax.swing.AbstractButton$Handler.actionPerformed(AbstractButton.java:2308)
	at java.desktop/javax.swing.DefaultButtonModel.fireActionPerformed(DefaultButtonModel.java:405)
	at java.desktop/javax.swing.DefaultButtonModel.setPressed(DefaultButtonModel.java:262)
	at java.desktop/javax.swing.AbstractButton.doClick(AbstractButton.java:369)
	at java.desktop/javax.swing.plaf.basic.BasicMenuItemUI.doClick(BasicMenuItemUI.java:1012)
	at java.desktop/javax.swing.plaf.basic.BasicMenuItemUI$Handler.mouseReleased(BasicMenuItemUI.java:1056)
	at java.desktop/java.awt.Component.processMouseEvent(Component.java:6614)
	at java.desktop/javax.swing.JComponent.processMouseEvent(JComponent.java:3342)
	at java.desktop/java.awt.Component.processEvent(Component.java:6379)
	at java.desktop/java.awt.Container.processEvent(Container.java:2263)
	at java.desktop/java.awt.Component.dispatchEventImpl(Component.java:4990)
	at java.desktop/java.awt.Container.dispatchEventImpl(Container.java:2321)
	at java.desktop/java.awt.Component.dispatchEvent(Component.java:4822)
	at java.desktop/java.awt.LightweightDispatcher.retargetMouseEvent(Container.java:4919)
	at java.desktop/java.awt.LightweightDispatcher.processMouseEvent(Container.java:4548)
	at java.desktop/java.awt.LightweightDispatcher.dispatchEvent(Container.java:4489)
	at java.desktop/java.awt.Container.dispatchEventImpl(Container.java:2307)
	at java.desktop/java.awt.Window.dispatchEventImpl(Window.java:2769)
	at java.desktop/java.awt.Component.dispatchEvent(Component.java:4822)
	at java.desktop/java.awt.EventQueue.dispatchEventImpl(EventQueue.java:772)
	at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:721)
	at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:715)
	at java.base/java.security.AccessController.doPrivileged(AccessController.java:391)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:85)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:95)
	at java.desktop/java.awt.EventQueue$5.run(EventQueue.java:745)
	at java.desktop/java.awt.EventQueue$5.run(EventQueue.java:743)
	at java.base/java.security.AccessController.doPrivileged(AccessController.java:391)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:85)
	at java.desktop/java.awt.EventQueue.dispatchEvent(EventQueue.java:742)
	at java.desktop/java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:203)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:124)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:113)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:109)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.desktop/java.awt.EventDispatchThread.run(EventDispatchThread.java:90)
`
Last year I did some modifications to the code so I was able to run with Apache Netbeans IDE 12.0 (only 1 warning left).
I made this code available today at:
[https://github.com/RaspCla/prm80-control-software/tree/main/control_software/](url)
This is the version with old RXTX driver and dll (I copied the dll to the lib directory), also.

One question: 
if I open the the old RXTXcomm.jar file I find much more classes I have in the new *.jar files, see attachement.
Do you see a possiblity that some of these addition
[image]
al classes are missing for my project?

alibabashack, 2021-04-11:
Hmm. As far as I can tell, the above Exception has nothing to do with rxtx. Seems to be some gui issue.

Yes, the new jar files only contain the serial port implementation. The old rxtx distribution had a lot of other experimental port types (parallel, I2C, etc.) in the making. But prm80 only uses the serial port, so that should be fine for you.

RaspCla, 2021-04-11:
I did one more test and added the "DriverManger..." line  also to the Option dialog.
I got a new erro message and now I feel the error is that the driver loader does not know my Windows10 operating system and so the dll is not loaded. 
Here the new error message:
`Apr. 11, 2021 6:31:12 PM gnu.io.LibraryLoader determineOsClass
INFO: The os.name value Windows 10 is unknown. Please file a bug.
Apr. 11, 2021 6:31:12 PM gnu.io.LibraryLoader load
SEVERE: Unknown operating system! Can't resolve library name
Exception in thread "AWT-EventQueue-0" java.lang.IllegalArgumentException: setSelectedIndex: 0 out of bounds
	at java.desktop/javax.swing.JComboBox.setSelectedIndex(JComboBox.java:622)
	at f4fez.prm80x0.gui.OptionsDialog.<init>(OptionsDialog.java:43)
	at f4fez.prm80x0.gui.PRM80X0View.optionMenuItemActionPerformed(PRM80X0View.java:774)
	at f4fez.prm80x0.gui.PRM80X0View$26.actionPerformed(PRM80X0View.java:578)
	at java.desktop/javax.swing.AbstractButton.fireActionPerformed(AbstractButton.java:1967)
	at java.desktop/javax.swing.AbstractButton$Handler.actionPerformed(AbstractButton.java:2308)
	at java.desktop/javax.swing.DefaultButtonModel.fireActionPerformed(DefaultButtonModel.java:405)
	at java.desktop/javax.swing.DefaultButtonModel.setPressed(DefaultButtonModel.java:262)
	at java.desktop/javax.swing.AbstractButton.doClick(AbstractButton.java:369)
	at java.desktop/javax.swing.plaf.basic.BasicMenuItemUI.doClick(BasicMenuItemUI.java:1012)
	at java.desktop/javax.swing.plaf.basic.BasicMenuItemUI$Handler.mouseReleased(BasicMenuItemUI.java:1056)
	at java.desktop/java.awt.Component.processMouseEvent(Component.java:6614)
	at java.desktop/javax.swing.JComponent.processMouseEvent(JComponent.java:3342)
	at java.desktop/java.awt.Component.processEvent(Component.java:6379)
	at java.desktop/java.awt.Container.processEvent(Container.java:2263)
	at java.desktop/java.awt.Component.dispatchEventImpl(Component.java:4990)
	at java.desktop/java.awt.Container.dispatchEventImpl(Container.java:2321)
	at java.desktop/java.awt.Component.dispatchEvent(Component.java:4822)
	at java.desktop/java.awt.LightweightDispatcher.retargetMouseEvent(Container.java:4919)
	at java.desktop/java.awt.LightweightDispatcher.processMouseEvent(Container.java:4548)
	at java.desktop/java.awt.LightweightDispatcher.dispatchEvent(Container.java:4489)
	at java.desktop/java.awt.Container.dispatchEventImpl(Container.java:2307)
	at java.desktop/java.awt.Window.dispatchEventImpl(Window.java:2769)
	at java.desktop/java.awt.Component.dispatchEvent(Component.java:4822)
	at java.desktop/java.awt.EventQueue.dispatchEventImpl(EventQueue.java:772)
	at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:721)
	at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:715)
	at java.base/java.security.AccessController.doPrivileged(AccessController.java:391)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:85)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:95)
	at java.desktop/java.awt.EventQueue$5.run(EventQueue.java:745)
	at java.desktop/java.awt.EventQueue$5.run(EventQueue.java:743)
	at java.base/java.security.AccessController.doPrivileged(AccessController.java:391)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:85)
	at java.desktop/java.awt.EventQueue.dispatchEvent(EventQueue.java:742)
	at java.desktop/java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:203)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:124)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:113)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:109)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.desktop/java.awt.EventDispatchThread.run(EventDispatchThread.java:90)
        DriverManager.getInstance().loadDrivers();
`

alibabashack, 2021-04-11:
aww, thank you. I opened issue #30

RaspCla, 2021-04-13:
Tests with new patch for #30 run ok, I'm able to 
- open the serial port
- send and and receive data
I still have issues with my project, seems they are not related to rxtx

I added the new libs to my project, you can find the current implementation 
[at Github](https://github.com/RaspCla/prm80-control-software/tree/main/control_software/)

EliasIEE, 2021-05-19:
> DriverManager.getInstance().loadDrivers();

Hi @alibabashack , is it necessary to import any package for this statement? The line (DriverManager.getInstance().loadDrivers();) is not accepted like this. Sorry I don't know much about Java, just trying to get an existing program to work with the new library version. Without the line the program compiles successfully, but it cannot open the serial port.

EliasIEE, 2021-05-19:
Since I do not get any further with the DriverManager, I copied the library into the Java directory:
`/usr/lib/jvm/jre/lib/amd64/librxtxSerial-linux-x86_64.so`

and set a symbolic link: `ln -s /usr/lib/jvm/jre/lib/amd64/librxtxSerial-linux-x86_64.so /usr/lib/jvm/jre/lib/amd64/librxtxSerial.so`

While trying to open the port I now get the following messages:

```
java.lang.UnsatisfiedLinkError: gnu.io.RXTXCommDriver.nativeGetVersion()Ljava/lang/String; thrown while loading gnu.io.RXTXCommDriv$

java.lang.NoClassDefFoundError: Could not initialize class gnu.io.RXTXCommDriver thrown while loading gnu.io.RXTXCommDriver
```

Do you have any idea how to solve this?

alibabashack, 2021-05-19:
@EliasIEE Yes, you will need to `import gnu.io.DriverManager;` Please also make sure you have added both rxtx jars to your classpath.

EliasIEE, 2021-05-19:
Thanks a lot, that seems to work without problems!