rxtx/rxtx #26: Usage Example

alibabashack, 2021-04-05:
We still need a good example how to use rxtx. This could be in the form of
* an additional artifact,
* a quick tutorial,
* (any other ideas welcome)

Johnlon, 2021-07-13:
Yes please.  Quite a blocker and unusual for a product not to have working example code and/or functional tests (eg using a loop back driver or something)

RaphaelWimmer, 2021-08-06:
Here's a minimal example that should work even if there are no serial ports on the machine:

```
import gnu.io.*;
import java.util.Enumeration;

public class ListPortsExample {

    public static void main(String[] args) {
        // the platform-specific driver needs to be loaded explicitly first!
        DriverManager.getInstance().loadDrivers();
        Enumeration ports = CommPortIdentifier.getPortIdentifiers();
        while (ports.hasMoreElements()) {
            CommPortIdentifier port = (CommPortIdentifier) ports.nextElement();
            System.out.println("found port: " + port.getName());
        }
    }
}
```

Compile with `javac -cp <path to rxtx-api jar> ListPortsExample.java` , e.g., `javac -cp lib/rxtx-api-2.2-stabilize-SNAPSHOT.jar ListPortsExample.java` 

Run with `java -cp <path to rxtx-api jar>:<path to rxtxSerial jar>:. ListPortsExample`, e.g., `java -cp lib/rxtx-api-2.2-stabilize-SNAPSHOT.jar:lib/rxtxSerial-2.2-stabilize-SNAPSHOT.jar:. ListPortsExample`

Example output (first two lines always logged by rxtx):

```
Aug 06, 2021 11:56:51 PM gnu.io.impl.serial.RXTXCommDriver loadRxtxProperties
INFO: No property file or resource found (using defaults)
found port: /dev/ttyUSB0
found port: /dev/ttyS0
```

If no output happens at all, the native driver has probably not been loaded.

@alibabashack - I'm not sure where to best place the example and have nearly no experience with Maven. Could you incorporate this example in the repo (maybe in rxtx-eamples/)?

migry, 2021-12-17:
I was looking for a simple example to show a crash caused by rxtxSerial.dll, so tried the above code. I got the following. Note: downloaded the jar and dll 64 bit from the fizzed repository (which appears to be no longer supported).

```
/cygdrive/g/MISC/SUMP/Test > javac -cp RXTXcomm.jar ListPortsExample.java
ListPortsExample.java:9: error: cannot find symbol
        DriverManager.getInstance().loadDrivers();
        ^
  symbol:   variable DriverManager
  location: class ListPortsExample
1 error
```

alibabashack, 2021-12-20:
Please try again with these the jars from https://github.com/alibabashack/rxtx/releases/tag/v2.2.1 they contain the loader you are using.

mecrowe, 2022-01-11:
@alibabashack The release link above is invalid.   It looks like there are no releases in your repo.

https://github.com/alibabashack/rxtx/releases

xyha1, 2023-03-24:
> @alibabashack The release link above is invalid. It looks like there are no releases in your repo.
> 
> https://github.com/alibabashack/rxtx/releases

have you got any links to rxtx?