rxtx/rxtx #33: Could not find artifact gnu.io.rxtx:...

mizapf, 2021-05-01:
I have a weird issue when building. On my normal machine, mvn compile runs cleanly. On another machine that I recently set up, mvn compile exits with the following output:

[ERROR] Failed to execute goal on project librxtxSerial-linux-x86_64: Could not resolve dependencies for project gnu.io.rxtx:librxtxSerial-linux-x86_64:so:2.2-stabilize-SNAPSHOT: Could not find artifact gnu.io.rxtx:rxtxSerial-native:zip:sources:2.2-stabilize-SNAPSHOT -> [Help 1]

I don't know what I'm doing differently here. On a third machine (installed as a VM) I have the same problem.

mizapf, 2021-05-04:
OK, I found the difference but I don't know how to fix it. 

The folder ~/.m2/repository/gnu is missing on my other machine. When I remove it on my primary machine, building RXTX also fails there. Did I forget to set up something?

alibabashack, 2021-05-15:
Could you please try again with "mvn install" instead?

Now, after merging #32 you will be provided with a -- hopefully useful -- error message which guides you to the selection of an appropriate binary platform:

e.g. mvn -Pwith-linux-x86_64 install

mizapf, 2021-05-16:
Yes, if you do it once with "mvn install", further invocations of "mvn compile" seem to work. Is that an intended behavior?

I used the compile operation because I am usually cautious about doing installs before I know what is happening. If I make something, I normally run "make -n install" to see where the program will be moved to before I actually let it install.

alibabashack, 2021-05-16:
Well, yes and no. Currently it is the intended behavior, because this was the only way I could make it work. Since we cannot build all native binaries on the same host (at least OSX needs to be build separately), I had to introduce a two step build process. This allows to install native binaries build on other hosts before assembling the final rxtxSerial.jar containing all available binaries. So the local maven repository is (miss-)used to aggregate the binaries.

I feel your concerns. Sadly, maven seems to live in the Java "compile once, run everywhere"-world and does not embrace the portability issues of native libs. That said, I have no experience with maven outside of this project. So I am happy to hear about different approaches.

mizapf, 2021-05-18:
As long as this is documented, and it is reproducible, this is something we can well live with.

kamarbaraka, 2023-06-06:
i tried injecting the gnu.io:rxtx dependency in my maven pom.xml file but i keep getting Could not find artifact gnu.io:rxtx:jar:2.1.7 in central