rxtx/rxtx #34: Problem if sender has RTS/CTS enabled on win64

huthomas, 2021-06-23:
If the sender device has RTS/CTS enabled, no data is received on win64.

The data can be received with the same code on linux 64bit, or on win32 with the rxtx lib version 2.1.7.

alibabashack, 2021-08-06:
Hi @huthomas ,
in order to investigate this issue a minimum example - or even better, a test case within the test package would be very helpful. Please also give a hint which hardware you are using.

huthomas, 2021-08-20:
Hi @alibabashack ,
thx for the reply. The problem occurred with laboratory devices for Elexis (https://github.com/elexis) users. We could reproduce it with a small Win terminal app (I only got an executable no link to the source) that allowed simulating RTS/CTS.
As this was critical for the users we switched to a different serial lib, sry.