rxtx/rxtx #36: Calls to CommPortIdentifier.getPortIdentifiers() returns immediately without any ports

RaphaelWimmer, 2021-07-27:
When calling `CommPortIdentifier.getPortIdentifiers()` from a simple example application, it immediately returns because `CommPortIndex` is `null` - even if there are accessible serial ports. Using `librxtx-java` from Debian instead works flawlessly.

It seems that the Java code never even accesses the native library (strace does not show any access, and no error is shown even if the library is missing). 
I have built the current library version both with OpenJDK 11 and 17. The problem appears on Debian Linux and probably also on Windows (at least we have the same symptoms there).

Is there an initialization step missing to fill the list of ports?

ZRunner, 2021-07-29:
I can repro with OpenJDK 14 on Ubuntu. From what I can tell it should even occur on any platform really.

Do you have an explicit tuto about installing and using `librtx-java`? I couldn't find any good one :/

RaphaelWimmer, 2021-07-29:
Oh, I found the solution hidden in https://github.com/rxtx/rxtx/issues/24#issuecomment-817287239 
> Then add the following line somewhere in your project. It must be executed once before an attempt to open a serial port is made:
DriverManager.getInstance().loadDrivers();
> This will extract and load the appropriate native library from the jar. There is no need to copy the dll manually.

Now it works.

@alibabashack : having to manually call `loadDrivers()` means that the new rxtx can not be used as a a drop-in replacement with legacy code (in my current case, rxtx is wrapped by another closed-source library). Would it be possible to check a variable `libraryLoaded` in each constructor and static method call, and call `loadDrivers()` automatically if needed?

alibabashack, 2021-08-06:
Hi @RaphaelWimmer ,
this is the intended "new" way, correct. Sorry you had to figure it out yourself. Now, that you made it work, please help documenting this usage. Feel free to take a look at https://github.com/rxtx/rxtx/issues/26 and simply write down simple instructions and provide them via a pull request. Other people will thank you! If nobody starts, things will stay hard. I cannot allocate any more time to this project, sorry. Working at this library has been very frustrating, because many people still seem to be using it, but only few try to help.

I admit the "new" version is not a drop-in replacement as you stated. The design decision was to load the native libs at a defined, explicit point in time, not during static init or as a side effect of object construction. Of course backward compatibility is very important. If you have an idea how to backport the behavior (maybe selectively) you are also welcome to suggest it via  pull request. Thanks!

RaphaelWimmer, 2021-08-06:
minimal example (list available ports) here: https://github.com/rxtx/rxtx/issues/26#issuecomment-894540555

RaphaelWimmer, 2021-08-06:
Thank you for the explanation (and reviving rxtx!)

> The design decision was to load the native libs at a defined, explicit point in time, not during static init or as a side effect of object construction

What is the advantage of doing this? I cannot think of a common use case where you would want to only load the native driver at a later point in time.

alibabashack, 2021-12-20:
Hi @RaphaelWimmer ,
sorry for the very long delay and thank you for providing the minimal example. I will try it incorporate it into documentation in the next days.

> What is the advantage of doing this? I cannot think of a common use case where you would want to only load the native driver at a later point in time.

Think about larger applications like a huge IDE or similar. I had the Netbeans Rich Client Platform in mind for example. If every lib would argue this way and load all its resources at startup, it would be a very slow experience. Libs should be designed for all use cases. For smaller applications it is simple to load the drivers right away. Larger applications should have the chance to load on use.

Regards Alex

alibabashack, 2021-12-20:
I am still thinking about providing a drop-in replacement backport for older applications, so I leave this ticket open.

pmguy, 2023-05-10:
Hi @alibabashack  - thank you for leaving this ticket open as a drop-in replacement for our old 32 bit rxtxSerial.dll and 2.2-debug.jar is just what we are looking for.
We maintain a mature application that has been running a 32 bit rxtxSerial.dll in a 32 bit JRE 1.8 but we have now to move that to OpenJDK 17 and therefore 64 bit. We were hoping that the DLL and jar available on [http://fizzed.com/oss/rxtx-for-java]() would do the trick but whatever we have tried results in an EXCEPTION_ACCESS_VIOLATION (see attached log dump
[hs_err_pid10936_redacted.log](https://github.com/rxtx/rxtx/files/11442421/hs_err_pid10936_redacted.log)).

I've read the comments about using DriverManager to preload the required drivers but that package does not exist in the rxtx jar from fizzed - I can see it here in github but not in a downloadable jar. And in reality we are only targetting Windows 10 so there is no real need to have any more native drivers (we include the approriate rxtxSerial.dll in the installation package).

Are you anywhere near providing this drop-in package please? I'm sure that we are not the only ones having this problem!
Pete