rxtx/rxtx #38: Fatal error

Karthik181007, 2021-07-30:
#
# A fatal error has been detected by the Java Runtime Environment:
#
#  EXCEPTION_ACCESS_VIOLATION (0xc0000005) at pc=0x0000000180005b00, pid=10380, tid=0x0000000000002524
#
# JRE version: Java(TM) SE Runtime Environment (8.0_291-b10) (build 1.8.0_291-b10)
# Java VM: Java HotSpot(TM) 64-Bit Server VM (25.291-b10 mixed mode windows-amd64 compressed oops)
# Problematic frame:
# C  [rxtxSerial.dll+0x5b00]
#
# Failed to write core dump. Minidumps are not enabled by default on client versions of Windows
#
# An error report file with more information is saved as:
# C:\RFID_Based_SmartShoppingSystem\hs_err_pid10380.log
#
# If you would like to submit a bug report, please visit:
#   http://bugreport.java.com/bugreport/crash.jsp
# The crash happened outside the Java Virtual Machine in native code.
# See problematic frame for where to report the bug.
#

Process finished with exit code 1

RaphaelWimmer, 2021-08-06:
You are most probably using the old rxtx library (~2008). 
Unfortunately, there is no new release yet.
However, you can build the new release from source.
There is no other solution for fixing this issue that I am aware of.

alibabashack, 2021-08-06:
Thank you @RaphaelWimmer  for helping out. @Karthik181007 please download the binaries "rxtx-distribution" from guthub actions CI:
https://github.com/rxtx/rxtx/actions/runs/844341221
You will need to add both contained jars to your project and initialize the library as described by @RaphaelWimmer  in https://github.com/rxtx/rxtx/issues/36

Karthik181007, 2021-08-07:
Actually, I downloaded lower version of java jdk 1.8 and it solved all this unnecessary errors.

Karthik181007, 2021-08-07:
Actually,I downloaded the lower version of java jdk 1.8 and it solved all the unnecessary errors .

closed 2021-08-07 by Karthik181007