rxtx/rxtx #39: Unknown packaging: dll @ line 6, column 17

guardiola98, 2021-12-17:
When i execute mvn -Pwith-windows-x86_64 install, i get this error:
[INFO] Scanning for projects...
[ERROR] [ERROR] Some problems were encountered while processing the POMs:
[ERROR] Unknown packaging: dll @ line 6, column 17
 @ 
[ERROR] The build could not read 1 project -> [Help 1]
[ERROR]   
[ERROR]   The project gnu.io.rxtx:rxtxSerial-windows-x86_64:2.2-stabilize-SNAPSHOT (C:\Users\jguardiolag\rxtx\rxtxSerial-windows-x86_64\pom.xml) has 1 error
[ERROR]     Unknown packaging: dll @ line 6, column 17
[ERROR] 
[ERROR] To see the full stack trace of the errors, re-run Maven with the -e switch.
[ERROR] Re-run Maven using the -X switch to enable full debug logging.
[ERROR] 
[ERROR] For more information about the errors and possible solutions, please read the following articles:
[ERROR] [Help 1] http://cwiki.apache.org/confluence/display/MAVEN/ProjectBuildingException

This is the line that throws that error:
[image]

I have tested by executing mvn -Pwith-windows-x86_64 compile as well as mvn -Pwith-windows-x86_64 clean install.
I would like to say that i am executing those commands from rxtx parent directory.

guardiola98, 2021-12-20:
Update:
I was doing the build on windows. I didn't know that the build does't work on windows. Now i have tested to make cross build on an ubuntu machine. I have done this steps:
sudo apt install default-jdk
sudo apt-get install build-essential
sudo apt-get install maven
sudo apt-get install gcc-mingw664-x86-64
git clone https://github.com/rxtx/rxtx.git -b development /path
cd path
mvn -Pwith-windows-z86-64 install

By making this i get this error:
[image]

Then, how has been said on #33 by @mizap and @alibabashack  , i have executed 
mvn -Pwith-windows-z86-64 install
mvn -Pwith-windows-z86-64 compile

When i compile, i don't have any error but i have the next problem:
How is said on #36 by @RaphaelWimmer (he found the solution on #24). I need two jar files:

- rxtx-api/target/rxtx-api-2.2-stabilize-SNAPSHOT.jar (contains the java API only)

- rxtxSerial/target/rxtxSerial-2.2-stabilize-SNAPSHOT.jar (contains the Java serial driver implementation and its native libraries) to the projects netbeans project.

But with the compile option, i only get the  rxtx-api/target/rxtx-api-2.2-stabilize-SNAPSHOT.jar

alibabashack, 2021-12-20:
Hi @guardiola98 ,
sadly it is rather difficult to project our use case onto the mave build system. I had to make changes to the build workflow again and the documentation is no longer up to date. Please take a look at the [github action workflow](https://github.com/rxtx/rxtx/blob/development/.github/workflows/rxtx-distribution.yml) for working build instructions or use the libs from https://github.com/alibabashack/rxtx/releases/tag/v2.2.1

guardiola98, 2021-12-24:
Hi, @alibabashack ,
First of all thank you for answer so quick. Having the two options that you gave to me, i have choosen using the libs from https://github.com/alibabashack/rxtx/releases/tag/v2.2.1 and it has worked perfectly for me. Thank you so much. I'm closing this issue.

closed 2021-12-24 by guardiola98