rxtx/rxtx #42: LibraryLoader: add support for Windows 11

fdeitelhoff, 2022-03-03:
I am working on a Java project with rxtx enabled to communicate to a device via the COM port.

On Windows 10 everything is working fine. On Windows 11 I get the following error:

März 03, 2022 3:09:23 PM gnu.io.LibraryLoader determineOsClass
INFORMATION: The os.name value Windows 11 is unknown. Please file a bug.
März 03, 2022 3:09:23 PM gnu.io.LibraryLoader load

This is understandable, because Windows 11 is not checked for in rxtx-api\src\main\java\gnu\io\LibraryLoader.java.

I try to create a PR to fix this.

VWrestle97, 2022-09-06:
Hi fdeitelhoff, 

I am working on a Java project with rxtx enabled to communicate to a device via the COM port.

For me, on Windows 8 everything is working fine. However, on Windows 10 I get the following error:

'.\SerialConnection.java:  error:  package [gnu.io](http://gnu.io/) does not exist'

Can you please provide me with your working rxtxSerial.dll file that works for Windows 10 64-bit and jdk1.8.0_321?

Thank you.

Oriana10, 2023-12-26:
Hello everyone, I'am getting the same error:

11:01:54 [32m INFO[m [35m11072[m --- [           main] [36mg.i.LibraryLoader                       [m : The os.name value Windows 11 is unknown. Please file a bug.
11:01:54 [31mERROR[m [35m11072[m --- [           main] [36mg.i.LibraryLoader                       [m : Unknown operating system! Can't resolve library name
11:01:54 [31m[ERROR][m [35m11072[m --- [           main] [36mc.j.a.InitializeInterfaces              [m - Line 53 : Error while configuring analyzer channel [Serial]